arduino-libraries/ArduinoBearSSL #17: ALPN support would be nice

sitase, 2020-02-08:
Needed for MQTT to AWS IoT per https://docs.aws.amazon.com/iot/latest/developerguide/protocols.html (if you have the default MQTTs port filtered out).

Support is in BearSSL, so it should just be a matter of exposing the possibility to configure.

thekunalsaini, 2020-03-20:
If you are connecting MQTT to AWS IoT then you can use the following :-

 According to AWS documentation, we have pubsub for connecting.
you can refer the documentation for further details.

Thank you

thekunalsaini, 2020-03-20:
@sitase @per1234

sitase, 2020-03-20:
@thekunalsaini Would you care to elaborate?

thekunalsaini, 2020-03-22:
@sitase Actually, if you check the documentation of AWS you will have pubsub which is used for publish and subscriber purpose and used to connect to Iot devices.

thekunalsaini, 2020-03-22:
@per1234

per1234, 2020-03-22:
@thekunalsaini why did you mention me? If there is some action you want me to take, then you need to provide a detailed explanation of that action and your reasoning for it.

You need to understand that busy people get an email every time you comment on an issue report or a pull request. This takes up time we could be using to do more important things. Productive input is very much welcome, but vague statements and mentions for no obvious purpose are not at all productive. I'm writing this reply when I have thousands of other things I could be doing with my time.

manchoz, 2020-03-30:
@thekunalsaini the low-level BearSSL library already has integrates the APLN support.

The low-level function you are looking for is [br_ssl_engine_set_protocol_names](https://github.com/arduino-libraries/ArduinoBearSSL/blob/f5674ef3eb638545dc60a935a7eebd41049575b0/src/bearssl/bearssl_ssl.h#L1323).

Please, consider reading the in-line APLN-related functions and flags documentation and implementing a corresponding extension method for the ArduinoBearSSL class. When done, please, submit it in a proper PR.

Thanks!

aentinger, 2020-05-18:
Reading through this I realise that the last information provided by @manchoz is actually relevant for the issue creator. @sitase Currently our hands are a bit full but if you could provide a good PR we can merge it in.