arduino-libraries/ArduinoBearSSL #30: sslClient.setInsecure(BearSSLClient::SNI::Insecure);

TheEngineer2020, 2020-06-01:
Good day,

We are developing an Arduino based device that sends an https post to a server. We use the Arduino MKR Zero and an ethernet shield. When we use this function:

  sslClient.setInsecure(BearSSLClient::SNI::Insecure);

We get this error when the compilation is done:

'class BearSSLClient' has no member named 'setInsecure'

Do you know what could be the origin of this error?

manchoz, 2020-06-01:
@TheEngineer2020 `setInsecure` is a brand new method and it is not yet included in the version of the library available on the Library Manager in the IDE or the CLI.

You need to install the repo version of the GitHub library.

PLEASE, be VERY careful when enabling this feature and don't forget to remove it while in production.

aentinger, 2020-06-02:
@TheEngineer2020 A new version of `ArduinoBearSSL` ([v1.5.0](https://github.com/arduino-libraries/ArduinoBearSSL/releases/tag/1.5.0)) has been release which contains this method. Other than that I can only echo @manchoz words ...
> PLEASE, be VERY careful when enabling this feature and don't forget to remove it while in production.

closed 2020-06-02 by aentinger

TheEngineer2020, 2020-06-02:
Thank you very much, you are awesome.