arduino-libraries/ArduinoBearSSL #37: Custom CA certificate

torntrousers, 2020-11-15:
I'm trying to use ArduinoBearSSL on an Arduino Nano 33 IoT board to talk to a server that requires mutual TLS, but can't get it to work. I'm guessing its because the server is using a custom certificate which ArduinoBearSSL doesn't know about.

I've tried using setInsecure(BearSSLClient::SNI::Insecure) but that I think must also disables presenting the client certificate thats set with sslClient.setEccSlot call, so the TLS connection still fails.

How can I  set a custom CA certificate? Is there any API to do that? 

TIA!

ffontaine, 2020-11-24:
You can provide your custom CA certificate thanks to Trusted Anchors:
`BearSSLClient sslClient(nbClient,TAs,1);`

TAs can be formatted from PEM file thanks to brssl tool (available from https://www.bearssl.org/git/BearSSL):
`brssl ta custom_ca.pem`

aentinger, 2020-11-24:
That's correct, thank you @ffontaine :rocket:

closed 2020-11-24 by aentinger

ffontaine, 2020-11-24:
It should also be noted that you must also provide the correct time through `ArduinoBearSSL.onGetTime` for the certificate validation (see the examples).