arduino-libraries/ArduinoBearSSL #41: Making ArduinoECCX08 optional ?

ffontaine, 2021-02-15:
I'm opening this issue to know if it would be possible to make ArduinoECCX08 optional?

Indeed, I'm currently using ArduinoBearSSL on an Arduino MKR NB 1500 but without using the ArduinoECCX08 (thanks to  `setEccVrfy`, `setEccSign`, etc.).
I'm planning to continue to use ArduinoBearSSL on a STM32 board so I made some modifications to protect all the ArduinoECC08 specific code by `ifndef ARDUINO_DISABLE_ECCX08`.
Ideally, I would like to upstream this modification but I'm not sure if it makes sense for you and I don't know if there is a standard way to declare or manage **optional** dependency in the Arduino world (i.e. if Arduino was autotools/cmake/meson based, I would declare an option such as --disable-eccx08).

ffontaine, 2021-04-07:
I made a proposal with #45

woodlist, 2021-04-11:
> I'm opening this issue to know if it would be possible to make ArduinoECCX08 optional?
>
> Indeed, I'm currently using ArduinoBearSSL on an Arduino MKR NB 1500 but without using the ArduinoECCX08 [...]

Please, explain how to get the ECCX08 optional, step by step, with example code. I'm planning to use this library in inegration with TinyGSM, on ESP32  platform.

aentinger, 2021-04-12:
This is possible due to merging #45. I suggest you read the discussion or take a look at the change itself, as it's fairly obvious.

closed 2021-04-12 by aentinger