arduino-libraries/ArduinoBearSSL #42: Works with 1.5.0, not with 1.6.0

torntrousers, 2021-03-12:
I'm using ArduinoBearSSL to connect to server with TLS and its works with the 1.5.0 release butnot with the 1.6.0 release. Going back through the commits trying each it breaks with this change:  https://github.com/arduino-libraries/ArduinoBearSSL/pull/31

The server does have a certificate with an ECDSA key.

Is there some bug, or am I suppose to explicitly call setEccVrfy/setEccSign now? And if so what would the arg be?

torntrousers, 2021-03-12:
It looks like the issue is with https://github.com/arduino-libraries/ArduinoBearSSL/blob/master/src/BearSSLClient.cpp#L50-L51 vs https://github.com/arduino-libraries/ArduinoBearSSL/blob/master/src/BearSSLClient.cpp#L240-L241

I'm not using mutual TLS for this call so don't call setEccSlot, so the ecVrfy and  _ecSign get those default values set at L50 and L51, which is not what they have in the 1.5.0 release.

torntrousers, 2021-03-12:
This change fixes it for me:
```
C:\cqtlibs\temp\demo-nano-33-iot\.pio\libdeps\nano_33_iot\ArduinoBearSSL>git diff
diff --git a/src/BearSSLClient.cpp b/src/BearSSLClient.cpp
index 67d00ee..905063e 100644
--- a/src/BearSSLClient.cpp
+++ b/src/BearSSLClient.cpp
@@ -46,8 +46,8 @@ BearSSLClient::BearSSLClient(Client* client, const br_x509_trust_anchor* myTAs,
   _numTAs(myNumTAs),
   _noSNI(false)
 {
-  _ecVrfy = br_ecdsa_vrfy_asn1_get_default();
-  _ecSign = br_ecdsa_sign_asn1_get_default();
+  _ecVrfy = eccX08_vrfy_asn1;
+  _ecSign = eccX08_sign_asn1;

   _ecKey.curve = 0;
   _ecKey.x = NULL;
```

What was the intention for using br_ecdsa_vrfy_asn1_get_default() and br_ecdsa_sign_asn1_get_default()?

ffontaine, 2021-03-13:
The goal was to be able to use ArduinoBearSSL without the ECC508 (e.g. using an IoT SAFE applet in a SIM card to sign and the main CPU to verify). To do so, I moved `eccX08_vrfy_asn1` and `eccX08_sign_asn1` to `setEccSlot`. You can set back the default values, I'll use `setEccVrfy` in my own code.

aentinger, 2021-04-07:
Fixed by merging #43.

closed 2021-04-07 by aentinger