arduino-libraries/ArduinoBearSSL #59: ArduinoBearSsl without eec

bvernham, 2022-01-12:
How do you use Bearssl without using a crypto chip.  How do you atrach the private, public, and certificate?

Thanks

aentinger, 2022-01-12:
I suggest reading the [BearSSL](https://bearssl.org/) documentation and/or poll the forum. A github issue is the wrong medium for support questions.

closed 2022-01-12 by aentinger

bvernham, 2022-01-12:
Well the problem is the example connecting to aws does not work or as the comments read works hit or miss and seem to be something to do with how bearssl accepts certificates.There does not seem to be a whole lot of support for this issue.  This is why i bouggt the board but can't even get a canned example to.work.

____________________________________________________________Top News - Sponsored By Newser3-Time Olympian Deon Lendore Dies in Car AccidentNorth Korea Launches 2nd Hypersonic Missile in a WeekLawmaker's Tweet About Wife Shoveling Snow Goes Viral