arduino-libraries/ArduinoBearSSL #60: failed to send a long data ( > 512bytes )

etakata, 2022-01-27:
Hello team,

I failed to send a long data(> 512bytes), using [BearSSLClient::write(const uint8_t *buf, size_t size)](https://github.com/arduino-libraries/ArduinoBearSSL/blob/92f83947bdfba67ff0c3fe99b801565dcf91b728/src/BearSSLClient.cpp#L109-L129)

Suppose that buf stored 600 bytes data and `br_sslio_write()` could send 512 bytes at maximum each call. 

Call `write(buf, 600)`

**Expected behavior**
`write()` sends 600 bytes and returns 600

**Current behavior**
`write()` sends 1024 bytes and returns 1024

Detail:
1. User calls `write(buf, 600)`
1. First time execution in the `while` loop
   * `br_sslio_write()` sends 512 bytes from `&buf[0]` and returns 512, buf points at &buf[512], written = 512
1. `written < size` is true
1. Second time execution in the `while` loop
   * `br_sslio_write()` sends 512 bytes from `&buf[512]` and returns 512, buf points at &buf[1024], written = 1024
1. `written < size` is false and escaping from the loop
1. `write()` returns 1024

To fix this issue, I'd like to change [here](https://github.com/arduino-libraries/ArduinoBearSSL/blob/master/src/BearSSLClient.cpp#L114)
from:  
`int result = br_sslio_write(&_ioc, buf, size);`
to:  
`int result = br_sslio_write(&_ioc, buf, size - written);`

aentinger, 2022-01-27:
Approved :+1: Can you please prepare a PR?

CC @pennam might be worth to integrate this change over at the ArduinoIoTCloud stack.

etakata, 2022-01-28:
@aentinger 

Thank you:)

I opened the pull request.-->https://github.com/arduino-libraries/ArduinoBearSSL/pull/61

aentinger, 2022-01-28:
Closing, due to merge of #61.

closed 2022-01-28 by aentinger