arduino-libraries/ArduinoBearSSL #65: No Documentation or Forum Support for libary

bvernham, 2022-05-03:
Besides the canned examples (in another libary) there is no documentation how to use the library or support in the forum.

BearSSL reference is not a 1 for 1 match for the functions in this library so it can not be readily referenced.

Some basic documentation or usable references would be helpful.

Thanks

Bruce

aentinger, 2022-05-04:
This has once been part of the Arduino IoT Cloud firmware stack but has since longer time been directly integrated with [ArduinoIoTCloud](https://github.com/arduino-libraries/ArduinoIoTCloud).

@alranel I'd consider marking this repo as unsupported and archiving this repositry, as already done e.g. for [ArduinoIoTCloudBearSSL](https://github.com/arduino-libraries/ArduinoIoTCloudBearSSL).

schnoberts1, 2024-07-19:
@aentinger , I note this repo continues to be updated and only 2 days ago was bumped with a new tag so it doesn't seem like it's abandonware as implied.