arduino-libraries/ArduinoBearSSL #68: A bad connection causes an inifinite loop (at least on an ESP32 but I think also on an Arduino)

Jeroen88, 2022-10-09:
The problem of #56 is even bigger: `::ClientRead()` and `::ClientWrite()` are called in bearssl ssl_io.c. If the socket client connection (`Client *c`) is lost or reset by the peer, both `c->read(buf, len)` and `c->write(buf, len)` will return a 0 (at least on the ESP32 but I think this is standard behavior also on an Arduino) thus causing an _infinite loop_ in `static int run_until(br_sslio_context *ctx, unsigned target)` in ssl_io.c.

In my situation this problem occurs on a bad WiFi connection, but I think it may also occur on a bad tcp/ip connection or if the peer closes the connection.

One solution would be to add a timeout in the `run_until()` function, but I think it is better not to touch the original library. So I made adaptations to `::ClientRead()` and `::ClientWrite()` that involve static variables, which I think is a bad programming habit, but this is the only way to record a state between calls to these functions. 
I did not introduce a new timeout value, but use the value of the socket client `c->getTimeout()`.

```
int BearSSLClient::clientRead(void *ctx, unsigned char *buf, size_t len)
{
  static bool notAvailableFlag = false;
  static uint32_t lastNotAvailableMillis;

  Client* c = (Client*)ctx;

  if (!c->connected() && !c->available()) {

    return -1;			// connection lost or closed by peer (in ssl_io.c low_read, which points to this function, fails on a -1. which it should if a connection is lost)
  }

  int result = c->read(buf, len);
  if (result == 0) {
    if(notAvailableFlag) {
      if(millis() - lastNotAvailableMillis > c->getTimeout()) {
        notAvailableFlag = false;  // for the next round

        return -1;		// timout read (in ssl_io.c low_read, which points to this function, fails on a -1. which it should if a connection is lost or closed by peer)
      }
    } else {
      notAvailableFlag = true; // First time no data available
      lastNotAvailableMillis = millis();
    }
  
    delay(10);			// Needed?
  } else {
    notAvailableFlag = false; // This flag was set but new data is available, so start again
  }

#ifdef DEBUGSERIAL
  DEBUGSERIAL.print("BearSSLClient::clientRead - ");
  DEBUGSERIAL.print(result);
  DEBUGSERIAL.print(" - ");  
  for (size_t i = 0; i < result; i++) {
    byte b = buf[i];

    if (b < 16) {
      DEBUGSERIAL.print("0");
    }
    DEBUGSERIAL.print(b, HEX);
  }
  DEBUGSERIAL.println();
#endif

  return result;
}

int BearSSLClient::clientWrite(void *ctx, const unsigned char *buf, size_t len)
{
  static bool notAvailableForWriteFlag = false;
  static uint32_t lastNotAvailableForWriteMillis;

  Client* c = (Client*)ctx;

  if (!c->connected()) {

    return -1;			// connection lost or closed by peer (in ssl_io.c low_write, which points to this function, fails on a -1. which it should if a connection is lost)
  }

  int result = c->write(buf, len);
  if (result == 0) {

    if(notAvailableForWriteFlag) {
      if(millis() - lastNotAvailableForWriteMillis > c->getTimeout()) {
        notAvailableForWriteFlag = false;  // for the next round

        return -1;		// timout write (in ssl_io.c low_write, which points to this function, fails on a -1. which it should if a connection is lost or closed by peer)
      }
    } else {
      notAvailableForWriteFlag = true; // First time impossible to write data to peer
      lastNotAvailableForWriteMillis = millis();
    }

    delay(10);			// Needed?
  } else {
    notAvailableForWriteFlag = false; // This flag was set but new data was written, so start again
  }

#ifdef DEBUGSERIAL
  DEBUGSERIAL.print("BearSSLClient::clientWrite - ");
  DEBUGSERIAL.print(len);
  DEBUGSERIAL.print(" - ");
  for (size_t i = 0; i < len; i++) {
    byte b = buf[i];

    if (b < 16) {
      DEBUGSERIAL.print("0");
    }
    DEBUGSERIAL.print(b, HEX);
  }
  DEBUGSERIAL.println();
#endif

  return result;
}
```

Jeroen88, 2022-10-09:
[Here](https://github.com/arduino-libraries/ArduinoBearSSL/blob/3cd7da18e201313358c198c23ccd25ec2ebff652/src/BearSSLClient.cpp#L492) also should be added: `if(errorCode() != BR_ERR_OK) return 0;` This will check for (among others) the connection being dropped during SSL negotiation.

jhochs, 2023-03-31:
This issue is causing poor performance of a program I'm running on a MKR 1500 to communicate via MQTT: about half the time when connecting to MQTT the program hangs indefinitely. I tried implementing @Jeroen88's clientRead() and clientWrite() but then it invariably won't connect at all. Any ideas why this is?

TMRh20, 2026-11-03:
I have been testing with a lossy wireless connection (nRF24L01+ radios via RF24Ethernet library) and seem to have hit the same problem.

One trigger for this hang seems to be: BearSSLClient::stop() calls br_sslio_close(), which waits for the peer's close_notify. Many
servers (mosquitto in my case) can fail to send one, so stop() blocks on an otherwise healthy connection. For example, arduino-mqtt calls
stop() whenever a connect times out, which may explain the MQTT hangs reported above. A read timeout limits this, but stop()
shouldn't wait for the peer at all: RFC 5246 §7.2.1 lets the side initiating the close close the connection without waiting for the
peer's close_notify. A workaround is to stop the underlying client first, so BearSSLClient skips the TLS close:

```cpp
EthernetClient ethClient;
class BearSSLClientQuickStop : public BearSSLClient {
public:
  BearSSLClientQuickStop(Client &net, const br_x509_trust_anchor *tas, int numTAs)
    : BearSSLClient(net, tas, numTAs), _net(net) {}
  void stop() override {
    _net.stop();
    BearSSLClient::stop();
  }
private:
  Client &_net;
};
// TLS layer on top of the RF24Ethernet TCP client, verified against TAs from secrets.h
BearSSLClientQuickStop sslClient(ethClient, TAs, (size_t)TAs_NUM);
```
  
Also, any read timeout should be separate from, or longer than, getTimeout()'s 1 s default. On slow links such as RF24, cellular or
poor WiFi, TCP retransmissions alone can exceed that during the handshake.

**Suggested fix** ( I ran this through Claude Code, and this is what it came up with)

In BearSSLClient::stop(), don't use br_sslio_close(). Instead:

1. Call br_ssl_engine_close().

2. Flush pending outgoing records with a bounded number of writes.

3. Call _client->stop() without reading anything.

Alternatively, put a timeout on the wait.