arduino-libraries/ArduinoBearSSL #71: Root certificates work for port 443, but not for port 888 (MQTTS)

tbobek, 2023-03-04:
uploading root certificate of `www.google.com:443`  on an Arduino MKR Wifi 1010 and connecting afterwards to it with the example `WifiSSLClient` works perfectly. 
But all my trials to connect via port 8883 TLS encrypted with a ca_certificate failed. I tried a local MQTTS-Broker and also the test.mosquitto.org broker. Uploading certificates onto the board for the domains 
- test.mosquitto.org:443
- test.mosquitt.org:8883
- mosquitto.org (which defaults to mosquitto.org:443) 

finished with no errors, but the connection fails. It shows a `-2` error code when asking with `MqttClient.connectError()` for it. The `MqttClient` object is derived from `BearSSLClient`. 

I also posted this issue in the arduino.cc forum: 
https://forum.arduino.cc/t/mqtts-with-arduino-mkr-wifi-1010/1094414

aentinger, 2023-03-20:
Not sure, but possibly an issue with [SNI](https://de.wikipedia.org/wiki/Server_Name_Indication)? Try using the [setInsecure](https://github.com/arduino-libraries/ArduinoBearSSL/blob/master/src/BearSSLClient.h#L76) API to see if it solves your problem, then fix the underlying SNI issue.

manchoz, 2024-05-24:
For future reference:

The full command is something like:

```sh
brssl ta my_tas_and_certs.pem > MyTAs.h
```

Then, add `MyTAs.h` file to the sketch folder/project and use the `TA` array in the constructor of the `BearSSLClient` object:

```C++
#include "MyTAs.h"

WiFiClient client;
BearSSLClient sslClient(client, TAs, TAs_NUM);
```

cbpure, 2024-06-05:
Same experience. I used ESP32S3 and SIM7670G and Azure iot hub. Bearssl works fine with port 443 towards to vsh.pp.ua:443 but when I tried to azure dps mqtts server with 8883 port, it does not work. returning -2 error(server connection error). It should work because Azure server is using Digicert Root G2 so there shouldn't be an issue. Therefore, I tried different SSL support from TinyGSM with SIMCOM A767X and it also has Digicert Root G2 and it works perfect. It seems ArduinoBearSSL mqtts support is not working over GSM client