arduino-libraries/ArduinoBearSSL #74: Error "SHA1Class SHA1" on an ESP32 when including HTTPClient.h

ksaye, 2023-10-11:
I am trying to use the HMACSHA256 signing capability on an ESP32 but I also need the HTTPClient.

When I include the HTTPClient.h, which depends on WiFiClientSecure.h in your sample, I get a redeclaration issue.

Sample code:

      /*
        ArduinoBearSSL SHA256
      
        This sketch demonstrates how to create a SHA256 hash and HMAC
        for an input string.
      
        This example code is in the public domain.
      */
      
      #include <HTTPClient.h>
      #include <ArduinoBearSSL.h>
      #include "SHA256.h"
      
      #ifdef ARDUINO_ARCH_MEGAAVR
      // Create the object
      SHA256Class SHA256;
      #endif
      
      void setup() {
        Serial.begin(9600);
        while (!Serial);
      
        // expect 0xE3B0C44298FC1C149AFBF4C8996FB92427AE41E4649B934CA495991B7852B855
        printSHA256("");
      
        // expect 0xD7A8FBB307D7809469CA9ABCB0082E4F8D5651E46D3CDB762D02D0BF37C9E592
        printSHA256("The quick brown fox jumps over the lazy dog");
      
        // expect 0xB613679A0814D9EC772F95D778C35FC5FF1697C493715653C6C712144292C5AD
        printHMACSHA256("", "");
      
        // expect 0xF7BC83F430538424B13298E6AA6FB143EF4D59A14946175997479DBC2D1A3CD8
        printHMACSHA256("key", "The quick brown fox jumps over the lazy dog");
      }
      
      void loop() {
      }
      
      void printSHA256(const char* str) {
        Serial.print("SHA256 of '");
        Serial.print(str);
        Serial.print("' is 0x");
      
        SHA256.beginHash();
        SHA256.print(str);
        SHA256.endHash();
      
        printResult();
      }
      
      void printHMACSHA256(const char* secret, const char* str) {
        Serial.print("HMAC-SHA256 of '");
        Serial.print(str);
        Serial.print("' with secret '");
        Serial.print(secret);
        Serial.print("' is 0x");
      
        SHA256.beginHmac(secret);
        SHA256.print(str);
        SHA256.endHmac();
      
        printResult();
      }
      
      void printResult()
      {
        while (SHA256.available()) {
          byte b = SHA256.read();
      
          if (b < 16) {
            Serial.print("0");
          }
      
          Serial.print(b, HEX);
        }
        Serial.println();
      }

Error:

      In file included from d:\Onedrive - Microsoft\Documents\Arduino\libraries\ArduinoBearSSL\src/ArduinoBearSSL.h:35,
                       from C:\Users\kevinsay\AppData\Local\Temp\.arduinoIDE-unsaved2023911-24336-1vv5l1k.d1vj\SHA256\SHA256.ino:11:
      d:\Onedrive - Microsoft\Documents\Arduino\libraries\ArduinoBearSSL\src/SHA1.h:50:18: error: 'SHA1Class SHA1' redeclared as different kind of symbol
       extern SHA1Class SHA1;
                        ^~~~
      In file included from C:\Users\kevinsay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/hal/include/hal/sha_types.h:25,
                       from C:\Users\kevinsay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/mbedtls/port/include/sha512_alt.h:28,
                       from C:\Users\kevinsay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/mbedtls/mbedtls/include/mbedtls/sha512.h:69,
                       from C:\Users\kevinsay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/mbedtls/mbedtls/include/mbedtls/entropy.h:34,
                       from C:\Users\kevinsay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFiClientSecure\src/ssl_client.h:11,
                       from C:\Users\kevinsay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFiClientSecure\src/WiFiClientSecure.h:26,
                       from C:\Users\kevinsay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\HTTPClient\src/HTTPClient.h:37,
                       from C:\Users\kevinsay\AppData\Local\Temp\.arduinoIDE-unsaved2023911-24336-1vv5l1k.d1vj\SHA256\SHA256.ino:10:
      C:\Users\kevinsay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32s3/include/esp_rom/include/esp32s3/rom/sha.h:25:5: note: previous declaration 'SHA_TYPE SHA1'
           SHA1 = 0,
           ^~~~
      
      exit status 1
      
      Compilation error: exit status 1

KevinNelSL, 2024-08-22:
@ksaye did you ever resolve this?

pennam, 2024-08-22:
@KevinNelSL you can try to use this PR https://github.com/arduino-libraries/ArduinoBearSSL/pull/89

elasticdotventures, 2024-09-06:
This fixed the issue for me.  Thanks!