arduino-libraries/ArduinoBearSSL #86: Question - trying to use with MKR1000 getting "Error compiling for board Arduino/Genuino MKR1000

IraSch, 2024-03-23:
I installed the BearSSL library latest version. I put the ArduinoBearSSLConfig.h file in the same directory as my sketch, and modified it to uncomment the    #define ARDUINO_DISABLE_ECCX08

In my sketch I have

#include "ArduinoBearSSLConfig.h"
#include <ArduinoBearSSL.h>

But I get a compile error: Error compiling for board Arduino/Genuino MKR1000

I also tried adding the #define at the top of my sketch, but still get the compile error. 

I thought I saw someone else using this library with MKR1000, so I must be missing something. Please advise. Thanks.

per1234, 2024-03-24:
Hi @IraSch. I'm going to ask you to post the full output from a compilation. Please do this:

1. Open the sketch that uses the **ArduinoBearSSL** library in Arduino IDE.
1. Select **Tools > Board > Arduino SAMD Boards (32-bits ARM Cortex-M0+) > Arduino MKR 1000 WiFi** from the Arduino IDE menus.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
1. Wait for the compilation to fail.
1. You will see a "**Compilation error: ...**" notification at the bottom right corner of the Arduino IDE window. Click the "**COPY ERROR MESSAGES**" button on that notification.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a comment here on the GitHub thread.
   In order to ensure the text is not corrupted, please use [code fencing](https://guides.github.com/features/mastering-markdown/).
1. Click the "**Comment**" button to post the output.

IraSch, 2024-03-24:
I am able to get my sketch to compile now. What I did was to put ArdruinoBearSSLConfig.h (with the DEFINE uncommented) in with the other Bear library files instead of having it in the same folder as my sketch. 

However when running the sketch I am no longer able to connect.  Using straight 

  client.connect("api.twilio.com, 443);

I am able to connect to Twilio's API - but Twilio then rejected my POST with a 400 error. So I thought that the MKR1000 may not be handling SSL fully and hit upon ArduinoBearSSL.  But when I try

   sslClient.connect("api.twilio.com, 443);

It immediately fails to connect. Is there something else I need to be doing? 

I also tried adding

   sslClient.setInsecure(BearSSLClient::SNI::Insecure);   

But that didn't help.

Also, unrelated, I've tried compiling a few of the example sketches and they have a problem with 

    return WiFi.getTime();  - Class 'WiFIClass' does not have a member named 'getTime'

Thank you.

IraSch, 2024-03-24:
Thank you for your help and for your library

per1234, 2024-03-24:
I suspected this was the problem. Unfortunately https://github.com/arduino-libraries/ArduinoBearSSL/pull/45 was not tested. The system of putting the `ArduinoBearSSLConfig.h` in the sketch folder as demonstrated in the library examples is fatally flawed because, at least in the official boards platforms (the situation is different for the 3rd party ["**STM32 MCU based boards**" platform](https://github.com/stm32duino/Arduino_Core_STM32) and perhaps some other 3rd party platforms), the sketch folder is intentionally not added to the the compiler's ["search path"](https://gcc.gnu.org/onlinedocs/cpp/Search-Path.html), so this conditional will always evaluate as `false`:

https://github.com/arduino-libraries/ArduinoBearSSL/blob/d7a68ae80871bfb94b8887b45df6145701ea3cd0/src/ArduinoBearSSL.h#L29

and thus an `ArduinoBearSSLConfig.h` in the sketch folder will have no effect on the library, as was previously reported three years ago at https://github.com/arduino-libraries/ArduinoBearSSL/pull/45#issuecomment-817744608, but never acted on.

In order to avoid other users suffering from the same poor experience caused by this defect, the pointless `ArduinoBearSSLConfig.h` files must be removed from the examples and the correct way to use the `ArduinoBearSSLConfig.h` system documented. We will use this issue to track that task so it must remain open even though you are no longer suffering from it.